meshtastic/Meshtastic-Apple #1061: 🐞 [Bug]: multiple admin keys kills the app

vinklat, 2025-01-17:
### Firmware Version

2.5.18

### What did you do?

I have hardware Heltec Wireless Tracker 1.1 with firmware 2.5.18
using cli I have set PKI keypair and two admin keys:
```
meshtastic \
  --set security.private_key "base64:someprivatekey....=" \
  --set security.public_key "base64:somepubkey....=" \
  --set security.admin_key "base64:someadminpubkey1...." \
  --set security.admin_key "base64:someadminpubkey2...."
```

### Expected Behavior

App should work.

### Current Behavior

After multiple admin keys set up - iPhone app connects the node, but disconnets BLE after a while. 
Finally app crashes (closes) after every BLE connect.
Factory reset config of node helped recovery.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-01-21:
You can set all three admin keys in the app

closed 2025-01-21 by garthvh

metrafonic, 2025-02-23:
This is still an issue, as the app crashes if the keys have been provisioned somewhere else. Its some great trial and error done by @vinklat to discover this, I have spent hours trying to find out what is wrong with my setup.

vinklat, 2025-02-24:
yes the problem is still there, I even think it does it even when restoring a node backup using cli. I won't report bugs to this project anymore, because probably nobody cares.