meshtastic/Meshtastic-Apple #1062: 🚀 [Feature Request]: iOS Shortcuts - Allow Send Message To NodeID

Xaositek, 2025-01-17:
### OS

iOS, iPadOS, macOS

### Description

Currently you can only send a message to a channel - this request is to allow it to permit DMs also.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-01-17:
I have this feature coded already. Just need to test it and send it to GitHub.

RCGV1, 2025-01-30:
The App Intents branch has the feature right now, untested

Xaositek, 2025-02-17:
Looks like this was pushed to TestFlight - Great work @RCGV1 and @garthvh ! Probably can close this one when we push it to production.

closed 2025-03-31 by garthvh