meshtastic/Meshtastic-Apple #1064: 🚀 [Feature Request]: keep message history when switching between different devices.

mikecarper, 2025-01-19:
### OS

iOS

### Description

All messages disappear when connecting to different devices. It'd be nice to not lose all that history.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

closed 2025-01-19 by garthvh

lemketron, 2025-05-11:
Is there any information available as to why this was closed as not planned as opposed to saving it as a backlogged feature request? That makes it sound like it has been outright rejected, despite two people taking the time to write up a request for it. 

In addition, I did search other open issues to see if someone else had requested it already, but didn't find it because it was closed rather than backlogged. 😕

garthvh, 2025-05-11:
Your impression is correct, it is not something that is being worked on at all and not planned, in fact new features the PKI for dm's make it even less likely than a year ago.