meshtastic/Meshtastic-Apple #1065: 🚀 [Feature Request]: In messages/maps be able to click on a node to get more information

mikecarper, 2025-01-19:
### OS

iOS

### Description

In the node list be able to DM the device from the node information page.  
On the map be able to get to the information page by clicking on the icon.  
On the messages page be able to get to the information page by clicking on the icon.  

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-01-19:
There is a details view from the map, and available from the node list

closed 2025-01-19 by garthvh

mikecarper, 2025-01-21:
[image]

I'd like to be able to DM/Trace Route/ etc from this  point in the map. Have a link that goes to the details found if you click on the node via the nodes top level menu. In the details be able to DM the node and not have to jump and find the node again.