meshtastic/Meshtastic-Apple #1075: 🐞 [Bug]: Direct Messages appear in iOS notifications, not visible in messages window

JFRHorton, 2025-01-27:
### Firmware Version

2.5.17

### What did you do?

iPhone 13, iOS 18.1.1, Meshtastic iOS app version 2.5.17

Connected to node in client mode, received/sent messages to another client node within range (same room) both via direct message and over a channel.

### Expected Behavior

Expect to see “chat” style message display in direct message window for node

### Current Behavior

When receiving a direct message, a notification is visible on the iOS notifications page, and the count of direct messages increases. However, when opening the direct messages page, none are visible.
Transmitting a direct message works as expected, except the transmitted message is not visible in the chat window.

Channel messaging behaves as expected.

[image]

[image]

[image]

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dylanbox, 2025-02-04:
Also encountering this

RCGV1, 2025-02-05:
@rcarteraz The user stated that the nodes are in the same room so probably not over MQTT.

AA7US, 2025-02-09:
I'm also experiencing this bug with latest iOS app and node firmware. It's happening when receiving DM's directly via RF (no MQTT is involved as mentioned in issue #1051).

vpnels, 2025-03-03:
Also seeing this. Where are the DMs? I can't find them. Clicking the notice takes me to the Contacts page. No DMs.

morria, 2025-03-20:
I frequently see this issue as well. It appeared to occur when I receive a DM message from someone before receiving their NodeInfo or User info.

If I'm reading this right, it looks like the `messageList` is populated by [this query](https://github.com/meshtastic/Meshtastic-Apple/blob/36d6664dfcff799e798970020423f02ebea03d93/Meshtastic/Extensions/CoreData/UserEntityExtension.swift#L18) which filters on

```swift
NSPredicate(format: "((toUser == %@) OR (fromUser == %@)) AND toUser != nil AND fromUser != nil AND isEmoji == false AND admin = false AND portNum != 10", self, self)
```

with `self` serializing with all available user info including `shortName` and `longName` as seen when adding a log line there as `NSLog("UserEntity: %@", self)`.

Should messages be keyed off of `nodeId` instead since we can only be guaranteed to have that nodeId for messages that appear before `NodeInfo`?

If we'd get away with that we could just change what we key messages on at insert and lookup time, or add a `description` value to the `UserInfo` extension that only includes the `id`.

```swift
       public override var description: String {
               return "User(id: \(id)"
       }
```

garthvh, 2025-03-20:
> I frequently see this issue as well. It appeared to occur when I receive a DM message from someone before receiving their NodeInfo or User info.
> 
> If I'm reading this right, it looks like the `messageList` is populated by [this query](https://github.com/meshtastic/Meshtastic-Apple/blob/36d6664dfcff799e798970020423f02ebea03d93/Meshtastic/Extensions/CoreData/UserEntityExtension.swift#L18) which filters on
> 
> NSPredicate(format: "((toUser == %@) OR (fromUser == %@)) AND toUser != nil AND fromUser != nil AND isEmoji == false AND admin = false AND portNum != 10", self, self)
> with `self` serializing with all available user info including `shortName` and `longName` as seen when adding a log line there as `NSLog("UserEntity: %@", self)`.
> 
> Should messages be keyed off of `nodeId` instead since we can only be guaranteed to have that nodeId for messages that appear before `NodeInfo`?
> 
> If we'd get away with that we could just change what we key messages on at insert and lookup time, or add a `description` value to the `UserInfo` extension that only includes the `id`.
> 
>        public override var description: String {
>                return "User(id: \(id)"
>        }

A node info and user is created from the nodeid automatically every time a packet is received from a node that is not in the node list.

garthvh, 2025-03-31:
Can't replicate this issue.

xe1brx, 2025-03-31:
I have the same problem.  LORA, no MTTQ.

GregIthaca, 2025-05-09:
We have six people sharing an MQTT server. There are no problems with the Android devices. There are no problems with public messages.
However, for all of us on iPhone, we see notifications for all incoming messages, but only **some** of the private messages are reflected in devices under the contacts. The other Direct Messages show completely blank screens. They do not even reflect the **outgoing** messages that are sent to that user, which seems like it might be helpful/indicative.
I have some screen shots I can share if that helps.

RCGV1, 2025-05-09:
Is this via MQTT? There's been some issues with sending dms via mqtt I think unrelated to this issue.

bconway, 2025-05-09:
> Is this via MQTT? There's been some issues with sending dms via mqtt I think unrelated to this issue.

In my case, no. MQTT is disabled everywhere.

GregIthaca, 2025-05-09:
I don't have a big enough local group to test it not over MQTT, so I can't say for sure if the issue appears for DMs in general. But it was very informative that we had both Android and Apple users on the call and could see how the problem was not replicable on Android.

RCGV1, 2025-05-09:
I heard someone mentioned that this happens when that node cycles out of your node db and when they send you a message it doesn't end up in messages screen. We need to narrow this issue down so that we can fix it, any logs (device or from iOS), screenshots, and screen recordings would be appreciated.

garthvh, 2025-05-09:
I think this was fixed in 2.6.1

closed 2025-05-09 by garthvh

GregIthaca, 2025-05-09:
I'm running 2.6.1, and another person who was seeing the same thing. I'm not sure how to get what logs you are looking for, but if you tell me where to look I'm happy to try to reproduce. In the meantime here is a screenshot of a message notification, and the (blank) message history behind it as soon as the notification went off-screen. You can also the red (5) notification bubble, but there is no way to clear it by looking at the (non-existent) messages.

[image]

[image]

bconway, 2025-05-12:
Still seeing this on iOS app 2.6.2. Notification comes in, the message list shows no new items in the list, and clicking on the notification takes me to an empty public channel. Screenshots attached.

[image]
[image]
[image]

bconway, 2025-07-13:
I've been able to gather some more data, as this is a continued problem on the latest iOS release. I have two nodes set up near each other with similar capabilities, one using the web client via wifi, and one using iOS via bluetooth. Both nodes are configured identically, other than the Heltec on wifi. From a segment of today:

Web client: 3 messages
iOS notifications: 3 messages (the same ones)
iOS chat: 2 messages (2 expected, one missing).

So it appears that the issue is not with the iOS-connected node, nor the messages being received by iOS. Can this be reopened? Thanks.

[image]

[image]

[image]

RCGV1, 2025-07-15:
Ok so after a ton of investigation it seems like this phantom message issue (what I'm dubbing it) happens on both ends for a node, sometimes it occurs when you send a message to a node and it actually get sent but never shows up in message history.

iceHtwoO, 2025-07-28:
Is there any fix for this issue?

tmulle, 2025-08-05:
Any fix? I still see this using both Lora and MQTT.
I am told I have new direct messages according to the count but when I click on the Direct message link I don't see the new messages. This is happening on both iOS and MacOS Meshtastic applications.

Clicking on Direct Messages doesn't do anything and I can't read my messages sent to me.

[image]

[image]

cvandesande, 2025-09-19:
Any reason why this was closed? It’s still an issue?