meshtastic/Meshtastic-Apple #1080: 🐞 [Bug]: Waypoints sometime create in the wrong location

tekstrand, 2025-01-28:
### Firmware Version

2.5

### What did you do?

1. Tap/hold to drop a waypoint

### Expected Behavior

Waypoint should be created at the location you tapped/held

### Current Behavior

Sometimes, the waypoint is dropped in california near San Jose. I can't figure out how to reliably reproduce it but you'll know it is happening if the distance is wrong during the creation process before save

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ponzano, 2025-01-28:
I just noticed this too, seems like every time the waypoint creation overlay is opened and you scroll a little bit, the coordinates get replaced 
Screen recording of the bug: https://share.icloud.com/photos/0fdxBdkYiRaUHzitvmsrJMlhQ

RCGV1, 2025-02-18:
I think the default position for waypoints is Apple Park Cupertino, so if the position is null it sets it there.

closed 2025-04-10 by github-actions[bot]