meshtastic/Meshtastic-Apple #1082: 🐞 [Bug]: Smart position sharing doesn't seem to update in the background

tekstrand, 2025-01-28:
### Firmware Version

2.5

### What did you do?

1. Enable both "Share Position" and "Smart Position" in Apple system settings
2. Allow Meshtastic "Always" access to your location in Apple system settings
3. Set node location/position config to Not Present but enable Smart Position with the default interval/distance
4. Keep the app "open" at least in the background, but close phone
5. Travel at least 100m (or minimum set distance)

### Expected Behavior

Position should continue to update even if the phone is locked/screen off

### Current Behavior

Position is not updated until opening the phone/app and acquiring a new position

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-31:
Apple will still kill the app in the background sometimes if your device is low on resources, the live activity is the best way to ensure that your app stays running in the background.

closed 2025-03-31 by garthvh

StrangePelaton, 2025-03-31:
> Apple will still kill the app in the background sometimes if your device is low on resources, the live activity is the best way to ensure that your app stays running in the background.

Live activity doesn't fix it either. It seemed to work when it was tried several weeks ago, but now it makes no difference. Two separate iphone users, identical issue. If the screen is locked, the location will not update. No settings were changed by either user, which I personally confirmed for both, between when it worked and when it stopped working.