meshtastic/Meshtastic-Apple #1084: 🚀 [Feature Request]: Notifications/telemetry do not persist if app is force closed

tekstrand, 2025-01-28:
### OS

iOS

### Description

I am not sure if this is possible with iOS but seems to work great on Android so figured I'd request parity.

Even after force closing the app, I'd like to remain connected to my node so that new notifications still come through, and shared location from my phone to my node continues to update.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-01-28:
Whe. You force close the app it is closed

closed 2025-01-28 by garthvh

tekstrand, 2025-01-28:
> When You force close the app it is closed

Did some more research into this and posting the information since it may be useful to another reader in the future.

Android supports persistent/background services for BLE connections, meaning that even if an app is force closed the Android device can remain connected and receive/send data. iOS does not support these things at this time.