meshtastic/Meshtastic-Apple #1086: 🐞 [Bug]: Nodes hop count filter

kruthoff, 2025-02-03:
### Firmware Version

2.5.18

### What did you do?

Use the hop count filter in the nodes list.

### Expected Behavior

Setting the filter to i.e. 3, I would expect to see all nodes which are max. 3 nodes away _including_ nodes without a hop.

### Current Behavior

Nodes with 0 hops are not visible anymore.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I'm not sure if this is a feature or a bug, but I'm not able to see nodes from 0 - 3 hops.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-03:
There is a seperate selection for Direct

closed 2025-02-03 by garthvh

victordomingos, 2025-11-01:
Actually, if the idea of the **Node Filters** is to filter by a specific number of hops (nor including those the are accessible with less than those hops), all the captions under the "**Hops Away**" slider that are like "_2 or less hops away_" are not correct. 

[image]

In my opinion, what would make more sense for the users would be to apply the filter as the captions indicate. For instance, in Portugal, the Meshtastic community has been strongly advocating for users to limit to 4 the "_Number of hops_"  in the LoRA settings.

Even the slider ruler itself implies that 0 to n hops are included in the filter, because there is a blue line on the left. 
And maybe "_All_" should be at the right end of the scale, since it includes all the other options, from Direct to 7 hops.