meshtastic/Meshtastic-Apple #1093: 🐞 [Bug]: Meshtastic App Says No Connection To Node, Node is connected in bluetooth and shows in app.

Castawhat, 2025-02-09:
### Firmware Version

2.5.4

### What did you do?

Pair heltec V3 to mac, use for 5 minutes, get on with hour of work weaving in and out of bluetooth zone, 

### Expected Behavior

node auto connects to device and app should realise that

### Current Behavior

[image]
app thinks node is and isnt connected

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

MacOS sequoia 15.2 
Macbook Pro (M1, 2020) 
App downloaded from app store

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

c0debabe, 2025-02-24:
I have seen this happen before.

xukashi, 2025-03-12:
same here with an iPhone SE 2020

closed 2025-04-19 by github-actions[bot]

albertusQuelle, 2025-07-31:
Same bug here, cant use meshtastic on iOS