meshtastic/Meshtastic-Apple #1095: 🐞 [Bug]: Portduino marked Unsupported

Xaositek, 2025-02-11:
### Firmware Version

2.5.18 (1355)

### What did you do?

Stood up some new Raspberry Pi radios with MeshAdv hats and noticed in all instances that iOS app marks them as "Unsupported".

Given that "Portduino" is really just our own code and platform agnostic, I don't know that there is a vendor to support the project and this is a carry over from the Web Flasher code as noted by @RCGV1 in Discord.

Can we move this into "Supported" to improve user experience?

### Expected Behavior

Portduino should be listed as Supported in iOS app

### Current Behavior

Portduino is listed as Unsupported in iOS app

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-17:
Portduino is not supported in the api

closed 2025-02-17 by garthvh

Xaositek, 2025-02-17:
@garthvh I realize this but it also looks a bit odd in the iOS app - maybe we should investigate a static override for this device?

garthvh, 2025-02-18:
The app uses the same approach as the webflasher, would need to be updated there.