meshtastic/Meshtastic-Apple #1100: 🐞 [Bug]: Node Name Updates Don't Reflect in App without Restarting

Xaositek, 2025-02-13:
### Firmware Version

2.5.18 (1355)

### What did you do?

I changed the "Owner" name is a node and noticed even after multiple hours the application doesn't reflect the update in my node list.

Old Name:
[image]

New Name:
[image]

### Expected Behavior

As NodeInfo updates are received by the radio, name updates are immediately reflected in the app

### Current Behavior

As NodeInfo updates are received by the radio, name updates are only updated if you force close the app and restart it.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

thebentern, 2025-02-13:
@Xaositek what happens if you disconnect / reconnect BLE? I'm wondering if I need to send the owner info back up over the PhoneAPI

Xaositek, 2025-02-13:
Confirmed a BLE disconnect / reconnect also works to fix it

Xaositek, 2025-02-13:
Also noticed owner short name experiences the same issue