meshtastic/Meshtastic-Apple #1101: 🚀 [Feature Request]: Add details to what changes before applying a QR code config

jedahan, 2025-02-15:
### OS

iOS

### Description

It would be nice to show more details of what will change by applying a config. I see a few increments, maybe each as separate pull requests.

1. Dump raw or decoded config on the screen
2. Pick the most important changes and display them better than a text dump, but allow full text dump expansion
3. Maybe add some icons for potentially 'dangerous' changes like power output

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [x] I am willing to submit a pull request for this feature.

### Additional comments

Came from observing a discussion in the discord a few days back

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-15:
This is in place already

jedahan, 2025-02-15:
Is there a pointer to where I can see that? I don't see it on `main` or `AppLinks` branches in SaveChannelQRCode.swift...

jedahan, 2025-02-15:
I think its #1097

garthvh, 2025-02-15:
Have you scanned a QR code on iOS?  It gives you a warning that the channels and lora config are going to be updated.

jedahan, 2025-02-15:
Yes, I was looking to enhance that screen to preview what settings would be changed, by parsing the payload. Instead of a generic warning.

garthvh, 2025-03-12:
The primary purpose of the QR code is to make it easy for devices sharing QR codes to communicate, happy to take a look at a method to more easily communicate what settings are going to be updated if it is not too confusing as QR codes are mostly used by new users.

closed 2025-04-19 by github-actions[bot]