meshtastic/Meshtastic-Apple #1105: 🐞 [Bug]: iOS app is very slow when the primary Channel has a lot of messages in its hisotry

AntoineMazuyer, 2025-02-17:
### Firmware Version

2.5.20.4c97351

### What did you do?

Hi,

I live in an area where we are a lot of users. Some day we exchange ~60 messages on the primary channel.

At some points, it is very difficult to navigate through the different tabs of the app (nodes, Bluetooth, etc…) and nearly impossible to scroll through the messages.

The only solution I found is to wipe the app data

### Expected Behavior

It should run fine 

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-02-17:
Having hundreds of unread messages creates a lot of activity when you are reading those messages.  Feel free to open a pull request if you have a solution for this.

closed 2025-02-17 by garthvh

AntoineMazuyer, 2025-02-19:
I don’t think you get my point.

The problem is not receiving a lot of messages or generating a lot of activity.

The problem is the app, even not connected to the radio, that is very laggy after one or two days of use

rfmoz, 2025-09-04:
As reported, scroll down is laggy, with a few new messages, turns the UX terrible.

What does the app need to do under the hood to display the messages? When it reports X unread messages, does it already have them in the phone memory, or does it need to fetch each one individually from the device? Maybe that explains the behavior.