meshtastic/Meshtastic-Apple #1115: 🐞 [Bug]:  traceroute notification via iOS

CosmicCEO, 2025-02-26:
### Firmware Version

2.5.20

### What did you do?

click iOS notification after traceroute request

### Expected Behavior

iOS traceroute notification: expect a link back to the target node in app, seems to link to whatever node was discovered last. 

### Current Behavior

iOS traceroute notification: seems to link to whatever node was discovered last. 

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-13:
@daviesgeek

daviesgeek, 2025-03-14:
@garthvh when I have a few min, I will see if I can get it done next week. I'm absolute slammed at work so I'm not getting a lot of extra time on stuff like this atm

garthvh, 2025-05-11:
Fixed in 2.6.3

closed 2025-05-11 by garthvh