meshtastic/Meshtastic-Apple #1121: 🐞 [Bug]: Lower Banner Doesn't Always Have Background Shown

Xaositek, 2025-03-03:
### Firmware Version

2.6.0

### What did you do?

Browsing the Node List will occassionally show fill in the background image, options are still clickable.

[image]

### Expected Behavior

Background is shown to have distinction between navigation and content.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-03:
This is a SwiftUI bug with long scroll views

Xaositek, 2025-04-29:
With recent TestFlight builds, I have not seen this behavior re-occur. I think you fixed it @RCGV1 !

closed 2025-04-29 by RCGV1