meshtastic/Meshtastic-Apple #1125: 🚀 [Feature Request]: Router: "Are you sure?"

shortwavesurfer2009, 2025-03-05:
### OS

iOS

### Description

It was recently brought up in https://github.com/meshtastic/Meshtastic-Android/issues/1611 The new people were creating routers without realizing that it could deteriorate the mesh for other nearby users and a function was added to make that slightly less likely.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Additional comments

I do not have an iOS device myself, so if this already exists, please disregard.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-03-05:
https://github.com/meshtastic/Meshtastic-Apple/pull/1124

shortwavesurfer2009, 2025-03-11:
Now both Android and Web have merged the functionality.

RCGV1, 2025-03-21:
Added in 2.5.21

closed 2025-03-21 by RCGV1

shortwavesurfer2009, 2025-03-21:
When might this go live? Apps.apple.com currently says 2.5.18 so it seems like it might be a while.