meshtastic/Meshtastic-Apple #1126: 🚀 [Feature Request]: Filter nodes on the map like you can filter in the nodes tab.

W0IFM, 2025-03-05:
### OS

iOS

### Description

I think it would be nice to be able to filter the nodes on the map like you can filter them in the nodes tab. It will be helpful when determining if Lora direct is possible. I do not know what a pull request is. lol

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

lemketron, 2025-05-10:
New to Meshtastic and iOS app, but would love to see this feature implemented (and others). Might be able to help out with PRs at some point down the road; in the mean time, what sort of sponsorship is needed to implement a feature?

pqhf5kd, 2025-05-24:
Would love to see this. Map is very ‘busy’ in my area and more than half the nodes are offline

austinhargis, 2026-02-13:
Just opened a PR for filtering "offline" nodes from the map. Beyond the filter for offline nodes, are there any other filters that would be "priority" for the Mesh Map?

PR: #1593

W0IFM, 2026-02-14:
Hi Austin!
I like the map filter in the Web Client, that was a nice add!

Other priorities for filters:
-RF Only with adjustable hop number from “direct” to 7
-MQTT only with adjustable hop number from “direct” to 7
-Days since seen, with adjustable day count from 1 to 60

And this is another thing to think about on the iOS app, a search window in the Remote Admin Node list.
Thanks!!

Wil Mannecke

Sent from [Proton Mail](https://proton.me/mail/home) for iOS.