meshtastic/Meshtastic-Apple #1128: 🚀 [Feature Request]: iTAK App support

niccellular, 2025-03-12:
### OS

iOS

### Description

Include support for iTAK in the Meshtastic iOS App.

Assume we can connect to a TCP port on localhost (127.0.0.1) that we will receive XML CoT from iTAK on and we will send XML CoT to iTAK.

### Participation

- [x] I am willing to pay to sponsor this feature.
- [x] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-03-14:
Are you currently working on this?

ThatOneGuyGreggers, 2025-03-17:
Commenting to show that there is interest in this feature

niccellular, 2025-03-17:
I have a partial solution basically the Meshtastic -> iTAK code is done, but the iTAK -> Meshtastic code is not written yet.

sk8er02, 2025-03-26:
following

jonathan-k4, 2025-03-27:
Do you think a background socket will get through Apples App Store review? My understanding is they will only give background process approval to apps like VoIP and VPN clients. Otherwise you may have to choose a background mode that would result in your socket being closed quite quickly.

Would be interested in exploring a VPN approach. Also wonder why iTak doesn’t just implement direct Bluetooth communication with the Meshtastic radio itself.

niccellular, 2025-04-02:
> Do you think a background socket will get through Apples App Store review? My understanding is they will only give background process approval to apps like VoIP and VPN clients. Otherwise you may [...]

I believe since Meshtastic iOS App already has mechanisms to persist in the background, we inherit that for free? Garth would know for sure.

RCGV1, 2025-04-07:
@garthvh I was asking the original poster of the issue as the "willing to submit PR" option was checked.

garthvh, 2025-04-07:
> [@garthvh](https://github.com/garthvh) I was asking the original poster of the issue as the "willing to submit PR" option was checked.

I know I asked him to submit it for tracking, will be app, tak and firmware code to complete it, needs 5k in sponsorship.

vch911, 2025-09-29:
> > [@garthvh](https://github.com/garthvh) I was asking the original poster of the issue as the "willing to submit PR" option was checked.
> 
> I know I asked him to submit it for tracking, will be app, tak and firmware code to complete it, needs 5k in sponsorship.

I'd pay for this feature as in-app purchased feature.