meshtastic/Meshtastic-Apple #1130: 🐞 [Bug]: Setting region on a factory reset AND disabling TX (same time) doesn't disable TX

TaukSnarkyAgrud, 2025-03-13:
### Firmware Version

2.5.20

### What did you do?

1. meshtastic --factory-reset
2. Bluetooth pair with device
3. Select "Set Region"
4. Set the region
5. Disable TX
6. Click Save

### Expected Behavior

TX is disabled

### Current Behavior

TX is enabled

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

If the nature of the development/device dictates that this is not possible, consider disabling controls that cannot be changed while setting region.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

thebentern, 2025-03-13:
I'm not sure this is a bug on the iOS app. I think we explicitly turn on tx enable when you set a LoRA region in the firmware

garthvh, 2025-03-13:
This is likely the firmware turning on tx when you go from unset (where tx is disabled by default) to a region

closed 2025-03-13 by garthvh