meshtastic/Meshtastic-Apple #1133: 🚀 [Feature Request]: Power graph should show charge curve

jasonfish, 2025-03-14:
### OS

iOS

### Description

Currently the node info graph shows discharge curve, but jumps to 100% when the device is plugged in. Showing the charge curve instead would be very useful for solar nodes.

Rather than drawing the blue line based on "power state," it should instead be derived from [volts / 4.2 = n%] , then plot. It appears that the voltage is already being calculated for use in the list below, it's just not being used for the blue line.

[image]

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-31:
The value changes to 101 when the device is plugged in, this is the expected behavior.

closed 2025-03-31 by garthvh

jasonfish, 2025-03-31:
Is there another way to see the charge state over time, such as for solar nodes?