meshtastic/Meshtastic-Apple #1138: 🚀 [Feature Request]: Support toggling UDP Multicast

vidplace7, 2025-03-18:
### OS

iOS

### Description

Please add support for toggling the new "UDP Multicast" feature added in
- https://github.com/meshtastic/firmware/pull/5779
- https://github.com/meshtastic/protobufs/pull/643

Currently can only be enabled via Python with this config:
```sh
meshtastic ... --set network.enabled_protocols 1
```

Should be a boolean value to enable/disable, I imagine.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

vidplace7, 2025-03-18:
CC @thebentern

garthvh, 2025-03-31:
Will be released with 2.5.21

closed 2025-03-31 by garthvh