meshtastic/Meshtastic-Apple #1147: 🐞 [Bug]: LoRa Config / Transmit Power does not offer 0 - this is the recommended setting

jschrempp, 2025-03-21:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.5.20

### What did you do?

Navigate to LoRa Config. Scroll down to Transmit Power. The currently selected dBm is displayed. The +/- buttons can be used to change the value. The value does not go to 0.

### Expected Behavior

0 is the recommended Transmit Power setting on the Meshtastic site. It specifies to use maximum device power. This should be a choice.

### Current Behavior

Transmit Power can decrement to 1.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

The documentation could change, but they closed my PR for that with a note that this is an Apple problem and I should open a bug here. Frankly, I like the Apple implementation but as a new user this was confusing. Perhaps an alternative is to add text to the Transmit Power section that says, "recommend to set this as high as possible".

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jschrempp, 2025-03-22:
Additional notes from Discord: 

NomDeTom 🇬🇧 — 4:07 AM
This is the real error. It should be capable of taking negative values. Unfortunately 0 equates to default, but -1 actually equates to output at the radio chip of -1. This was also an error in android, corrected by #1629

garthvh, 2025-03-31:
Negative power is a bad UX experience, this should be corrected in the firmware if it is necessary.

closed 2025-03-31 by garthvh