meshtastic/Meshtastic-Apple #1151: 🚀 [Feature Request]: add support for the new board MeshLink

ponzano, 2025-03-26:
### OS

iPadOS, iOS, macOS

### Description

Since the MeshLink board is now officially supported by the firmware and present in protobuf, it would be nice if the app actually showed it correctly instead of what it does now.
If needed I can provide more info about the board, I’ll attach a png image in the meantime.
Thank you 

[image]

[image]

### Participation

- [x] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benkyd, 2025-03-27:
as @qwerty_812 on the discord pointed out. MESHLINK also is missing on the firmware

ponzano, 2025-03-27:
> as @qwerty_812 on the discord pointed out. MESHLINKs enumeration is also is missing on the firmware
> 
> https://github.com/meshtastic/firmware/blob/6c17694b64a391b06bded312f063d0e480380a81/src/platform/nrf52/architecture.h#L49

Oh, I'll try to make a PR to add that then

ponzano, 2025-03-27:
Nevermind, my colleague already did

https://github.com/meshtastic/firmware/pull/6404

garthvh, 2025-03-27:
This data is provided by the device hardware API, it is not available to be included in the apps yet

closed 2025-03-27 by garthvh