meshtastic/Meshtastic-Apple #1156: 🐞 [Bug]: Messages Filters do not when switching between tabs

wingcomm, 2025-03-30:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.3.640e731

### What did you do?

After specifying filters on the Messages view, they are no longer applied when you switch to another view like Bluetooth or Nodes and then back to Messages.

### Expected Behavior

Filters should always be applied as set by the users no matter what tab they switch to...

### Current Behavior

Filters are not applied correctly until you hit the filter button and toggle the filter you want on/off.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-03-31:
I can't replicate this issue, unless you close the app the filters stay in place when you come back, it does re-run the filter as more nodes can come in while you are in other tabs

wingcomm, 2025-03-31:
> I can't replicate this issue, unless you close the app the filters stay in place when you come back, it does re-run the filter as more nodes can come in while you are in other tabs

Yes, that's what I would expect to occur, but the filters are not working reliably for me at all. What would help show what I'm experiencing?

tannerellen, 2025-04-01:
I'm experiencing this bug as well. Happens very consistently for me. Here is a video.

https://github.com/user-attachments/assets/ec0971fd-595a-4b23-956e-bdd463ff849e

wingcomm, 2025-04-02:
@tannerellen Thanks for sharing. Nodes list seems to squirrelly after tapping around in Settings...

tannerellen, 2025-04-07:
@wingcomm Yep, can confirm I have run into the same issue with the nodes list now too.

RCGV1, 2025-05-08:
@garthvh I have identified the issue: the initialized default pedicate overrides the .onApppear searchUserList() predicate opperation. Adding a short delay to the .onAppear lets it fire after the init stage and it works as intended. This is not a clean solution but I have not played much with Core Data predicates so looking for your advice. 

Heres the default values:
```swift
@FetchRequest(
		sortDescriptors: [NSSortDescriptor(key: "lastMessage", ascending: false),
						  NSSortDescriptor(key: "userNode.favorite", ascending: false),
						  NSSortDescriptor(key: "pkiEncrypted", ascending: false),
						  NSSortDescriptor(key: "userNode.lastHeard", ascending: false),
						  NSSortDescriptor(key: "longName", ascending: true)],
		predicate: NSPredicate(
		  format: "userNode.ignored == false && longName != '' AND NOT (userNode.viaMqtt == YES AND userNode.hopsAway > 0)"
		), animation: .default
	)
	var users: FetchedResults<UserEntity>
```

garthvh, 2025-06-11:
Fixed in 2.64 App Store release

closed 2025-06-11 by garthvh