meshtastic/Meshtastic-Apple #1168: 🐞 [Bug]: SecureInput View Fixes

garthvh, 2025-04-02:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

All

### What did you do?

Clicked the eye on a key value using the secure input control in the security config view under settings

### Expected Behavior

Make the toggle for the custom SecureInput view only toggle the control you are working with.

### Current Behavior

All the Secure inputs on the view are toggled

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-04-03:
@garthvh Should public key not be a secure input as only the private key is a secret?

garthvh, 2025-04-03:
> [@garthvh](https://github.com/garthvh) Should public key not be a secure input as only the private key is a secret?

It was recent updated and is just a read only text control now.

closed 2025-04-05 by RCGV1