meshtastic/Meshtastic-Apple #1169: 🐞 [Bug]: Meshtastic app on iOS loses new messages after tapping on the notification.

wingcomm, 2025-04-03:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.4.b89355f

### What did you do?

I have observed my LiliyGo T-Echo on 2.5 and 2.6 is not delivering all direct or channel messages received by the meshtastic radio. I'll receive a new message, get a iOS Push notification of the new message with a preview, but it will not be in my channels or messages when I open the app. However, it will still be on the screen of my LilyGo T-Echo...

### Expected Behavior

All messages received by the radio should appear in the app.

### Current Behavior

Some messages are delivered to the app but are lost permanently even though they remain on the screen of my T-Echo.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Scrazzled, 2025-04-04:
I'm also losing messages on the iOS app. Not sure if it's quite the same thing, but any messages that get populated in the app are lost after BT disconnect whether they had been read by me or not. Case in point, I connected to a radio, I saw that there were 7 messages, BT disconnected, after reconnecting all messages were removed before I was able to open the messages and read them.

wingcomm, 2025-04-06:
@Scrazzled I'm still trying to narrow this behavior down as well. My unit is still connected to the radio when messages are lost when I tap on the push notification.

educationfutures, 2025-04-27:
I'm having the same issue, but probably better described as what @Scrazzled is experiencing. In my case, I have a private channel for three devices, one is a T-Deck, and the other are Heltec v3 and T1000-E radios. The Heltec and T1000 are connected to a Mac and iOS tablet, respectively. When I take the T-Deck on the road to test range, 85% of messages that are acknowledged as received by the LoRA devices are not displaying in the Meshtastic apps despite both the Mac and tablet being connected to the radios at all times.

tekstrand, 2025-04-28:
I am also losing messages.

bplein, 2025-04-28:
In our local mesh we have been speculating that the lost iOS messages may be the result of the messages being a reply or Tapback to a message that our node didn't see. 

In other words, it doesn't know how to connect it as a reply to an unknown message so it discards it.

RCGV1, 2025-05-08:
@wingcomm Does this only happen to emoji tapback messages or all messages?

Geordon, 2025-05-10:
I appear to be having this issue as well. One of my local mesh-mates has sent me messages on a couple of different days.  On my iPhone, I saw one message in the locked-screen notification as it flashed past the screen unlocking.  I went to look for the message again and it was nowhere to be found.

Is there any way to retrieve messages from the node directly, as in if I were to connect to the serial port via USB on my Mac?  Or does the message get cleared somehow?

closed 2025-05-11 by garthvh

bconway, 2025-05-12:
Still seeing this on iOS app 2.6.2. Notification comes in, the message list shows no new items in the list, and clicking on the notification takes me to an empty public channel. Screenshots attached. Thanks!

[image]
[image]
[image]

tekstrand, 2025-07-13:
@garthvh I think this is still happening. Specifically seems to be reactions/tapbacks to messages where you didn't get the message.

wingcomm, 2025-07-13:
Still seeing this a well...

zwa-vault, 2025-08-03:
@bconway: thanks for the screenshots. Exactly the same behaviour on my side. Push notification is shown, but channel is empty. Seems to be a difficult on e for the devs :-/

zwa-vault, 2025-08-03:
@devs: please reopen the issue!

KingXerxes512, 2025-09-04:
Just experienced this issue on version 2.6.17. Please reopen this issue.

versj, 2025-09-15:
Same here sadly. When clicking on a push notification I can see the messages from my mates. Impossible to access them any other way

zwa-vault, 2025-09-15:
Same here! Keep getting push notifications, but some are nowhere to be found in the channel. This issue has never been solved!

versj, 2025-09-15:
> Same here sadly. When clicking on a push notification I can see the messages from my mates. Impossible to access them any other way

“Fixed” it by pressing the “clear app data” button.