meshtastic/Meshtastic-Apple #1171: 🐞 [Bug]: Bluetooth connection issues with newest iOS app version - repeatedly retries, unexpected BT connection drops

Scrazzled, 2025-04-04:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.5.20

### What did you do?

With this app version (2.5.22(1400)), attempt to connect to meshtastic radios through the app.  
Observe the connection trying and failing. 

### Expected Behavior

Bluetooth connection is normal within normal ranges. I don't expect to experience repeated BT connection failures every time I attempt to connect to my radios. 

### Current Behavior

BT connection handling is very bad with this version of the iOS app (version 2.5.22(1400)). 
I've been using Heltec T114 (firmware 2.5.20) for a while, and a T-Deck (firmware 2.5.18) and the bluetooth connection is equally bad. Within a normal range (~25' in this case), I'm experiencing failures to connect initially for a few minutes and once it finally connects (without immediately disconnecting) the BT connection drops after a few minutes anyway. 
This doesn't appear to impact connections at very close ranges (within a few inches) but my use case requires more distance than that (but still within normal BT range, which wasn't a problem for the past couple of iOS app versions). 

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Impact: This renders my meshtastic radios effectively useless if I can't connect reliably to them in the locations I need them to be. 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-04-04:
There are zero changes to the Bluetooth handling

RCGV1, 2025-04-04:
Try forgetting the node in Bluetooth settings of your iPhone settings

garthvh, 2025-04-07:
Not seeing any BLE issues.

Git-Hub-u-s-e-r, 2025-10-28:
Same Problem Here - Heltec Wireless Tracker V1.1 with Firmware 2.6.11.60ec05e - iOS App V. 2.7.5

After a successful connection, the device disconnects after a few minutes with the error "too many retries." Sometimes turning the device off and on helps, but most of the time the Bluetooth connection on the iPhone has to be deleted and re-established. Then the whole cycle starts over again.

walldodge, 2026-03-10:
Still cant connect to nrf52 after it being online for 1-2 days. Connection fails on Retrieving nodes on every retry. Android phone connects successefully. Reseting the node, phone app or bluetooth pairing does not work.