meshtastic/Meshtastic-Apple #1185: 🐞 [Bug]: App crashes when reconnecting to same device

tapomme, 2025-04-24:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.5.20

### What did you do?

Hi,
The app crashes everytime it tries to reconnect to the same device (IOS and MacOS) in BT.
I have to connect to another device and then connect back  to the previous one.
Same problem when trying with T-Echo, T1000-E or Heltec V3.
Same probleme after rebooting all Apple and Meshtastic devices .

IOS 18.3.2
MacOS  Sonoma 14.6.1

### Expected Behavior

App not crashing

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-04-24:
I need some logs or something to take a look at this, otherwise there is not enough info here to pin down a cause.

Are you using a standard config and firmware?

tapomme, 2025-04-25:
Everything is standard (app, firmwares etc ..)
I don't know where to find logs on iphone about this problem.
Am I alone with this problem ?

tapomme, 2025-04-25:
All I have is i n MacOS:

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [92659]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.5.18 (1355)
App Item ID:           1586432531
App External ID:       872191172
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               503

Date/Time:             2025-04-25 10:54:56.7421 +0200
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        5B941BF0-A81C-CD51-4EF1-871BCC668708

Sleep/Wake UUID:       BC568A96-FE74-487B-BDE3-7A4059CA5936

Time Awake Since Boot: 610000 seconds
Time Since Wake:       1123 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000102f02c54

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [92659]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Meshtastic                    	       0x102f02c54 0x102d64000 + 1698900
1   Meshtastic                    	       0x102fb5c9c 0x102d64000 + 2432156
2   Meshtastic                    	       0x102e0c3a8 0x102d64000 + 689064
3   Meshtastic                    	       0x102e104e8 0x102d64000 + 705768
4   CoreBluetooth                 	       0x1a9140468 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 208
5   CoreBluetooth                 	       0x1a91405ac -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 104
6   CoreBluetooth                 	       0x1a913cce4 -[CBPeripheral handleMsg:args:] + 616
7   CoreBluetooth                 	       0x1a9119ad0 -[CBCentralManager handleMsg:args:] + 168
8   CoreBluetooth                 	       0x1a91074b4 -[CBManager xpcConnectionDidReceiveMsg:args:] + 192
9   CoreBluetooth                 	       0x1a91073ac __30-[CBXpcConnection _handleMsg:]_block_invoke + 52
10  libdispatch.dylib             	       0x193ffb750 _dispatch_call_block_and_release + 32
11  libdispatch.dylib             	       0x193ffd3e8 _dispatch_client_callout + 20
12  libdispatch.dylib             	       0x194004a14 _dispatch_lane_serial_drain + 748
13  libdispatch.dylib             	       0x194005578 _dispatch_lane_invoke + 432
14  libdispatch.dylib             	       0x19400bac8 _dispatch_main_queue_drain + 748
15  libdispatch.dylib             	       0x19400b7cc _dispatch_main_queue_callback_4CF + 44
16  CoreFoundation                	       0x1942cead4 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
17  CoreFoundation                	       0x19428c258 __CFRunLoopRun + 1996
18  CoreFoundation                	       0x19428b434 CFRunLoopRunSpecific + 608
19  HIToolbox                     	       0x19ea3519c RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x19ea34fd8 ReceiveNextEventCommon + 648
21  HIToolbox                     	       0x19ea34d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x197aeacc8 _DPSNextEvent + 660
23  AppKit                        	       0x1982e14d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
24  AppKit                        	       0x197addffc -[NSApplication run] + 476
25  AppKit                        	       0x197ab5240 NSApplicationMain + 880
26  AppKit                        	       0x197d08654 _NSApplicationMainWithInfoDictionary + 24
27  UIKitMacHelper                	       0x1ad569f50 UINSApplicationMain + 972
28  UIKitCore                     	       0x1c37417bc UIApplicationMain + 148
29  SwiftUI                       	       0x1dd6f44e0 0x1dc726000 + 16573664
30  SwiftUI                       	       0x1dd6f4328 0x1dc726000 + 16573224
31  SwiftUI                       	       0x1dd328608 0x1dc726000 + 12592648
32  Meshtastic                    	       0x103049724 0x102d64000 + 3036964
[... 384 lines cut ...]
  "logWritingSignature" : "7dd2c1eb097391bd99f3775eb8ebc868b1117b53",
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "64b21a7351cbb02ce3442e4e",
      "factorPackIds" : {
        "REMINDERS_GROCERY" : "6647f0f7b6a75d3dc32993e7"
      },
      "deploymentId" : 240000042
    },
    {
      "rolloutId" : "6425c75e4327780c10cc4252",
      "factorPackIds" : {
        "SIRI_HOME_AUTOMATION_INTENT_SELECTION_CACHE" : "642600a457e7664b1698eb32"
      },
      "deploymentId" : 240000004
    }
  ],
  "experiments" : [

  ]
}
}

`

tapomme, 2025-04-25:
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Meshtastic [60132]
Path:                  /Applications/Meshtastic.app/Contents/MacOS/Meshtastic
Identifier:            gvh.MeshtasticClient
Version:               2.5.18 (1355)
App Item ID:           1586432531
App External ID:       872191172
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               503

Date/Time:             2025-04-25 10:58:23.1036 +0200
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        5B941BF0-A81C-CD51-4EF1-871BCC668708

Sleep/Wake UUID:       BC568A96-FE74-487B-BDE3-7A4059CA5936

Time Awake Since Boot: 610000 seconds
Time Since Wake:       1329 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000102eaec54

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [60132]

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Meshtastic                    	       0x102eaec54 0x102d10000 + 1698900
1   Meshtastic                    	       0x102f61c9c 0x102d10000 + 2432156
2   Meshtastic                    	       0x102db83a8 0x102d10000 + 689064
3   Meshtastic                    	       0x102dbc4e8 0x102d10000 + 705768
4   CoreBluetooth                 	       0x1a9140468 -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 208
5   CoreBluetooth                 	       0x1a91405ac -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 104
6   CoreBluetooth                 	       0x1a913cce4 -[CBPeripheral handleMsg:args:] + 616
7   CoreBluetooth                 	       0x1a9119ad0 -[CBCentralManager handleMsg:args:] + 168
8   CoreBluetooth                 	       0x1a91074b4 -[CBManager xpcConnectionDidReceiveMsg:args:] + 192
9   CoreBluetooth                 	       0x1a91073ac __30-[CBXpcConnection _handleMsg:]_block_invoke + 52
10  libdispatch.dylib             	       0x193ffb750 _dispatch_call_block_and_release + 32
11  libdispatch.dylib             	       0x193ffd3e8 _dispatch_client_callout + 20
12  libdispatch.dylib             	       0x194004a14 _dispatch_lane_serial_drain + 748
13  libdispatch.dylib             	       0x194005578 _dispatch_lane_invoke + 432
14  libdispatch.dylib             	       0x19400bac8 _dispatch_main_queue_drain + 748
15  libdispatch.dylib             	       0x19400b7cc _dispatch_main_queue_callback_4CF + 44
16  CoreFoundation                	       0x1942cead4 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
17  CoreFoundation                	       0x19428c258 __CFRunLoopRun + 1996
18  CoreFoundation                	       0x19428b434 CFRunLoopRunSpecific + 608
19  HIToolbox                     	       0x19ea3519c RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x19ea34fd8 ReceiveNextEventCommon + 648
21  HIToolbox                     	       0x19ea34d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
22  AppKit                        	       0x197aeacc8 _DPSNextEvent + 660
23  AppKit                        	       0x1982e14d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
24  AppKit                        	       0x197addffc -[NSApplication run] + 476
25  AppKit                        	       0x197ab5240 NSApplicationMain + 880
26  AppKit                        	       0x197d08654 _NSApplicationMainWithInfoDictionary + 24
27  UIKitMacHelper                	       0x1ad569f50 UINSApplicationMain + 972
28  UIKitCore                     	       0x1c37417bc UIApplicationMain + 148
29  SwiftUI                       	       0x1dd6f44e0 0x1dc726000 + 16573664
30  SwiftUI                       	       0x1dd6f4328 0x1dc726000 + 16573224
31  SwiftUI                       	       0x1dd328608 0x1dc726000 + 12592648
32  Meshtastic                    	       0x102ff5724 0x102d10000 + 3036964
33  dyld                          	       0x193e23154 start + 2476

[... 475 lines cut ...]

Model: MacBookAir10,1, BootROM 10151.140.19, proc 8:4:4 processors, 16 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Display: DELL P2715Q, 6016 x 3384, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Jan 13 2024 03:06:59 version 18.20.440.9.7.8.167 FWID 01-7be24857
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en6
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB3.1 Hub
USB Device: Belkin Pro Dock Display
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Parade Tech Billboard
USB Device: Jabra SPEAK 510 USB
USB Device: USB Billboard Device
USB Device: Universal USB-C 8-in-1 Dual Display Core Hub
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
`

tapomme, 2025-05-05:
Are those informations relevant ?
Thanks

garthvh, 2025-05-11:
I can't replicated this and don't have any other reports of this crash, I would recommend factory reseting the device in question,.

closed 2025-05-11 by garthvh

tapomme, 2025-05-12:
Hi,

I have the same problem with all meshtastic devices ! and the problem exists with IOS and MacOS app.

So i don't know what device should I reset !
o.p.

tapomme, 2025-05-13:
Same problem with Ipad OS 18.3.2 
Tried firmwares 2.6.4 on devices ... same problem again