meshtastic/Meshtastic-Apple #1193: 🐞 [Bug]: Italian translation error

ponzano, 2025-04-28:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.2

### What did you do?

>Open nodes list
>nodes that are charging are reported as “DEPARTMENT OF DEFENCE” (?????)
>get confused
>report the translation error

[image]

[image]

### Expected Behavior

“Charging” should be translated as “In carica”

### Current Behavior

Translation says “Dipartimento della difesa”

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

It was a funny error 😂

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

lollo03, 2025-04-29:
I created a MR for this #1194

garthvh, 2025-04-29:
Will be merged in 2.6.0

closed 2025-04-29 by garthvh