meshtastic/Meshtastic-Apple #1205: 🐞 [Bug]: Node Map Shows on the node details view from the mesh map allowing for node map inception

garthvh, 2025-05-06:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

ALL

### What did you do?

Clicked on the circle for node info popup on the node map from the mesh map.

### Expected Behavior

We should hide the node map if the parent view is the mesh map

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-05-06:
It’s not a bug but a feature

RCGV1, 2025-05-07:
There is a good way to tell what the parent view is in Swift UI. Plus the user has to sit there and open a ton of tabs purpusfully.