meshtastic/Meshtastic-Apple #1206: 🐞 [Bug]: macOS app slow - video included

rinchen, 2025-05-06:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

N/A

### What did you do?

The macOS app continues to be very slow and experiences beachballs. This has been an ongoing issue for a long while. Restarting the app seems to make it go a little faster but after leaving it running for a day or two causes the app to slow down dramatically taking 60+ seconds to execute a command (like switch a tab or type a character in chat). 

I made a video with the mac version and the app version displayed so you can understand.  I've updated the various connected devices' firmware multiple times as well. Note that the iOS app does not have this problem, only the macOS one.

[https://e.pcloud.link/publink/show?code=XZM1MqZcPGyQgOc9yXWsk4nKQvNSyILyXUy](https://e.pcloud.link/publink/show?code=XZM1MqZcPGyQgOc9yXWsk4nKQvNSyILyXUy)

### Expected Behavior

macOS should be as responsive as the iOS app

### Current Behavior

[https://e.pcloud.link/publink/show?code=XZM1MqZcPGyQgOc9yXWsk4nKQvNSyILyXUy](https://e.pcloud.link/publink/show?code=XZM1MqZcPGyQgOc9yXWsk4nKQvNSyILyXUy)

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

rinchen, 2025-05-06:
[image]
[image]

Open Files
`/Users/joey/Library/Containers/gvh.MeshtasticClient/Data
/Applications/Meshtastic.app/Contents/MacOS/Meshtastic
/usr/lib/libobjc-trampolines.dylib
/Library/Preferences/Logging/.plist-cache.0rToel0n
/Users/joey/Library/Containers/gvh.MeshtasticClient/Data/Library/Application Support/Meshtastic/Meshtastic.sqlite-shm
/private/var/db/analyticsd/events.allowlist
/System/Library/CoreServices/SystemVersion.bundle/English.lproj/SystemVersion.strings
/usr/lib/dyld
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-6291460-v2.csstore
/usr/share/icu/icudt76l.dat
/private/var/db/timezone/tz/2025b.1.0/icutz/icutz44l.dat
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/SystemAppearance.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/Aqua.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/FauxVibrantLight.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/VibrantLight.car
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/DarkAqua.car
/System/iOSSupport/System/Library/PrivateFrameworks/UIKitCore.framework/Versions/A/Resources/Localizable.loctable
/System/iOSSupport/System/Library/Frameworks/SwiftUI.framework/Versions/A/Resources/MainMenu.loctable
/System/Library/PrivateFrameworks/SFSymbols.framework/Versions/A/Resources/CoreGlyphs.bundle/Contents/Resources/Assets.car
/System/Library/PrivateFrameworks/SFSymbols.framework/Versions/A/Resources/CoreGlyphsPrivate.bundle/Contents/Resources/Assets.car
/System/iOSSupport/System/Library/PrivateFrameworks/UIKitCore.framework/Versions/A/Resources/Artwork.bundle/Contents/Resources/Assets.car
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/MenuCommands.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/InputManager.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/DictationManager.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/WritingTools.loctable
/System/Library/Extensions/AGXMetalG15X_M1.bundle/Contents/Resources/ds.g15s
/System/Library/Fonts/SFNS.ttf
/System/Library/Caches/com.apple.IntlDataCache.le.kbdx
/System/Library/Keyboard Layouts/AppleKeyboardLayouts.bundle/Contents/Resources/AppleKeyboardLayouts-L.dat
/System/Library/Keyboard Layouts/AppleKeyboardLayouts.bundle/Contents/Resources/InfoPlist.loctable
/System/Library/Extensions/AGXMetalG15X_M1.bundle/Contents/MacOS/AGXMetalG15X_M1
/System/Library/PrivateFrameworks/CoreUI.framework/Versions/A/Resources/DesignLibrary-macOS.bundle/Contents/Resources/macOSRepositories/LightStandard.car
/System/Library/Fonts/Helvetica.ttc
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/Common.loctable
/System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Resources/Localizable.loctable
/System/Library/PrivateFrameworks/Shortcut.framework/Versions/A/Resources/SearchManager2.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/Placeholders.loctable
/Users/joey/Library/Containers/gvh.MeshtasticClient/Data/Library/Application Support/.tipkit/tips-store.db-shm
/Applications/Meshtastic.app/Contents/Resources/Assets.car
/System/Library/LinguisticData/en/US/hyphenation.dat
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/16777235_419/functions.data
/System/Library/PrivateFrameworks/RenderBox.framework/Versions/A/Resources/archive.metallib
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/32024/libraries.data
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/Menus.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/Accessibility.loctable
/System/Library/Fonts/Keyboard.ttf
/System/iOSSupport/System/Library/Frameworks/MapKit.framework/Versions/A/Resources/Localizable.loctable
/System/Library/Fonts/SFCompact.ttf
/System/Library/PrivateFrameworks/CoreUI.framework/Versions/A/Resources/DesignLibrary-macOS.bundle/Contents/Resources/macOSRepositories/LightFauxVibrantStandard.car
/System/iOSSupport/System/Library/Frameworks/MapKit.framework/Versions/A/Resources/Assets.car
/System/Library/Fonts/SFNSRounded.ttf
/System/Library/Fonts/Menlo.ttc
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/FauxVibrantLightGraphite.car
/System/Library/LinguisticData/Latn/Dict2.dat
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/InfoPanel.loctable
/System/Library/Frameworks/AppKit.framework/Versions/C/Resources/Ruler.loctable
/System/Library/Fonts/Apple Color Emoji.ttc
/Library/Caches/com.apple.iconservices.store/664D07F3-56F7-34DD-B6D3-701893ED1102.isdata
/Library/Caches/com.apple.iconservices.store/7BED83CA-5772-38F0-BCC8-965D0C65C7F1.isdata
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/com.apple.iconservices/store.index
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/Graphite.car
/System/Library/PrivateFrameworks/CoreNLP.framework/Versions/A/Resources/tokruleLE.data
/System/Library/LinguisticData/Latn/Dict1.dat
/System/Library/PrivateFrameworks/DataDetectorsCore.framework/Versions/A/Resources/com.apple.datadetectorscore.cache.urlifier.system
/System/Library/Fonts/Supplemental/Galvji.ttc
/System/Library/CoreServices/SystemAppearance.bundle/Contents/Resources/VibrantLightGraphite.car
[... cut ...]
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/16777235_419/functions.data
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/16777235_419/functions.list
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/32024/libraries.data
/private/var/folders/5v/dy8xbb392nj0l5bznppdphvc0000gn/C/gvh.MeshtasticClient/gvh.MeshtasticClient/com.apple.metal/32024/libraries.list
->0xd0714141b4888887
/Users/joey/Library/Caches/GeoServices
`

rinchen, 2025-05-06:
Process Sampling

[Sample of Meshtastic.txt](https://github.com/user-attachments/files/20071014/Sample.of.Meshtastic.txt)

garthvh, 2025-05-06:
A catalyst app is never going to be as responsive as the main iOS app, it exists mostly because it is easy to publish from the iOS codebase without a lot of modifications.

rinchen, 2025-05-10:
[Meshtastic_2025-05-10-115558_whiterun.cpu_resource.diag.txt](https://github.com/user-attachments/files/20140871/Meshtastic_2025-05-10-115558_whiterun.cpu_resource.diag.txt)

garthvh, 2025-05-11:
How much ram does your MacBook Pro have?

garthvh, 2025-05-11:
Not a lot I can do about catalyst issues, keeping your Mac up to date would be my reccomendation.

closed 2025-05-11 by garthvh

rinchen, 2025-05-11:
I read that a few years ago there was a catalyst performance issue when you use an external (or multiple) displays. That's not the situation in my case.

rinchen, 2025-05-11:
I went through the Catalyst developer documentation and dev forms and there are suggestions, but not an exact match, that say 1) check the build information in Xcode and 2) follow the macOS optimization recommendations webpage for Catalyst.

rinchen, 2025-05-11:
I tried to build this using Xcode as a macos target vs iPad target but xcode is being xcode and won't let me do it without signing up for the developer account. I also seem to be missing swiftlint.

garthvh, 2025-05-11:
All those recommendations are for interface optimizations like using buttons instead of swipe gestures.

rinchen, 2025-05-11:
I've made some progress.  I completely uninstalled the app and reinstalled it from the store. It wiped my entire history of 1200 nodes, channel messages from years, etc.  My guess is that the app has difficulty handling that amount of data.

rinchen, 2025-05-16:
@garthvh are there any plans to address what appears to be the root cause as indicated in my previous comment?