meshtastic/Meshtastic-Apple #1220: 🐞 [Bug]: Tapping Nodes in node detail view should return to node list

lemketron, 2025-05-11:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.4.b89355f

### What did you do?

From the node list tap on a node for node details view
Now tap the Nodes button at the bottom of the screen
This should return (pop) back to Nodes list view.

### Expected Behavior

Tapping Nodes (or tapping Messages in Message view) should pop/return to previous view.

### Current Behavior

Tapping Nodes does nothing and forces user to tap on the “back” arrow in the upper left part of the screen.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

Tapping settings also returns to previous view so user may expect the other buttons to also do this.

Channels button should also pop (return) to Channel list from a channel message list.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-05-11:
This is not how the swift ui navigation control manages this, all of the navigation is managed from the top bar.

lemketron, 2025-05-11:
Then why does it work that way (tapping the bottom icon goes “back”) for Settings? And tapping Nodes when in the list view seems to show all even when filtered to something less than all. But tapping Nodes in detail does nothing. Hard to imagine it’s not fixable. 

I’m getting the feeling that usability suggestions are not welcome in this project though I hope that’s not true. There are so many things that could be done to make this app more useful (and friendly to use), but I won’t bother writing them up if they’re not wanted.

garthvh, 2025-05-11:
Settings is the only two page view in the app

garthvh, 2026-05-17:
This is still broken in the current codebase. `NodeList` uses a `NavigationSplitView` controlled by `router.selectedNodeNum`. A `popToRoot(tab: .nodes)` function exists in `Router.swift` that sets `selectedNodeNum = nil`, but it is never called anywhere — so tapping the Nodes tab while viewing a node detail does nothing.

The fix is to wire up tab-reselection to call `router.popToRoot(tab: .nodes)`. On iOS 18 this could use the new `Tab` API's built-in reselection behavior; on iOS 17 it requires an explicit `onChange` on the tab selection binding that detects a same-tab tap and calls `popToRoot`.

closed 2026-05-17 by garthvh