meshtastic/Meshtastic-Apple #1223: 🐞 [Bug]: App launch destroys TipKit data store causing tips to show on every app launch

wojo, 2025-05-13:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

n/a

### What did you do?

Launch the app, dismiss tips from views (e.g. on the Messages view), quit the app and re-launch the app.

### Expected Behavior

Tips should not appear again.

### Current Behavior

Tips appear and must be dismissed manually.

### Participation

- [x] I am willing to submit a pull request for this issue.

### Additional comments

Looks like debug code may have been left in that calls `Tips.resetDatastore()` on `MeshtasticAppleApp.init()` which completely resets the data store for TipKit.

https://github.com/meshtastic/Meshtastic-Apple/commit/ebbe7c2dccb55f54ce1097afb509762461d9b3e5

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-05-13:
This is intended, users dismissed the tips and then complained that they did not have the information provided by the tip kit, it shows on over launch now.

wojo, 2025-05-13:
Interesting, was that the majority of users? Very odd for TipKit to be used, able to be dismissed, and then constantly pop back up.

Some ideas:

1. Add a "Hide Tips" (or expert mode) in the settings
2. If tips should be shown always, can remove TipKit and just show a panel with hints/etc that always display. I could try a few different stylings to perhaps make it less obtrusive in that situation.

Thoughts?