meshtastic/Meshtastic-Apple #1229: 🐞 [Bug]: Cannot see direct messages

goyaweb, 2025-05-17:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.2

### What did you do?

First time user.
Send a direct message to a node next to me. 
Receive a message back.

### Expected Behavior

message sent can be seen, message received can be seen.

### Current Behavior

As soon as I hit send my message disapears but arrives as recipient.
When I receive an answer I get a notification but when opening the direct message the screen is empty again.

[image]

Strange other thing is that when I go out of range of my own node or turn it off I get an extra Contact in my direct message list with the name of my node. When opening this contact I can then see the whole conversation but can't see with whom it was. The contact i was having the conversation with remains having only an empty screen.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Whatnoww, 2025-05-19:
Having this particular issue.

Adding some context that this seems to happen when a node is first seen locally and then over mqtt. I have two sensecap t1000e where ocasionally I will have the node on iOS show up with user data, then a while later it'll also show up as a node without a key exchange. Can't reproduce this behavior on Android with the same node.

Whatnoww, 2025-05-19:
Yep, as suspected, this code here seems to apply universally and seems to be the root cause: https://github.com/meshtastic/Meshtastic-Apple/pull/814/commits/045bd7124af1a58bfa68efadf7fd16abce2951d7

Not sure why this was applied universally to all mqtt servers. My private one should not have the iOS app enforcing a 0 hop limit by removing a contact from the chatlist.

razorseal, 2025-05-20:
Having the same issue. However I am getting notifications showing that message came. Also the chat bubble at bottom right shows the notification and there's no way to erase it out. Only solution is to reinstall.  Running latest apk and ios 

[image]
[image]

RCGV1, 2025-05-24:
@goyaweb @razorseal @Whatnoww What iOS app version were you on when this occurred?

razorseal, 2025-05-24:
iOS 18.5 for me

RCGV1, 2025-05-24:
@razorseal The Meshtastic app version not iOS version

razorseal, 2025-05-24:
Oops sorry about that. I’m on 2.6.2 (1421)

RCGV1, 2025-05-24:
Can you guys capture logs when this happens by going to settings (bottom) logs

luu176, 2025-05-26:
Is this bug being fixed? I can't see any messages with some people

RCGV1, 2025-05-27:
@luu176 Thanks for the logs, can you also display debug logs

luu176, 2025-05-27:
aren't the logs I sent debug logs? it says debug logs

RCGV1, 2025-05-30:
@Dimonstar do the nodes have a locked or unlocked icon next to them in the node list or messaging list?

RCGV1, 2025-05-30:
Never mind I see it in the screenshot

luu176, 2025-05-31:
so is there a fix?

mangelajo, 2025-06-17:
I have also been hitting this sometimes, only an app db reset will fix it

va3jme, 2025-07-14:
I have had this issue, under a very specific condition.

- Add a new node to my favorites list
- Send and receive between me and new node, no problem
- Re-flash new-node, but do NOT update key in new node to be the same as it was before, in other words, it has a different key
- Send messages from me to new node, they will be accepted
- New node sees that a message came in, but it is not showing in the DM

The fix is to delete the new node from your system... possibly even reset your node-db. Then on the "new node" you download the key that was used when you created the node the first time. Now things should work fine. This is why you should always back up your key.... if you ever need to re-flash, and you want everyone to still be able to reach you, you will need to use the same key you always used.

Hope that is clear

Tested with three different phone (2 iOS, 1 android) and three Seeedstudio SenseCap T1000E units. Multiple tests confirm for me anyway that this fix works.

garthvh, 2025-07-16:
As of 2.6.12 he notifications for emoji have been updated so that they link to the message they are attached to, unmessagable nodes that have sent or received messages are now visible and there is no MQTT hops based filtering anymore.

closed 2025-07-16 by garthvh

holgerscherer, 2025-07-25:
bug is still in macos app ver 2.6.15
can't tell about who is the sender because i did not get notifications.

holgerscherer, 2025-07-25:
[image]

go-bears, 2025-08-03:
I'm seeing this same bug on ios 18.5 and firmware 2.7.1