meshtastic/Meshtastic-Apple #1240: 🐞 [Bug]: Bluetooth does not reconnect automatically after going out of range

Lozagon1234, 2025-06-01:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.4.b89355f

### What did you do?

On iOS, the Meshtastic app does not automatically reconnect to the node via Bluetooth after the device goes out of range and then returns. This results in missed messages unless the user manually opens the app to re-establish the connection.

Steps to reproduce:

1.     Connect the iPhone to a node via Bluetooth (e.g., Heltec V3 or T-Echo T114).

2.     Walk away or otherwise move the phone out of Bluetooth range.

3.     Return to Bluetooth range.

4.     Wait — the phone does not reconnect automatically.

5.     Open the Meshtastic app — connection is re-established immediately.

### Expected Behavior

The iOS app should automatically reconnect to the node once the phone is back within Bluetooth range, without needing to manually open the app.

### Current Behavior

No automatic reconnection happens unless the app is manually opened. This prevents the user from receiving messages until they remember to open the app again.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Additional comments

I don’t currently have other boards to test if the issue is limited to these models or if it’s general to all.
Devices tested:

    iPhone (iOS)

    Heltec V3 node

    T-Echo T114 node

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-06-02:
Your device is not advertising anymore if it does not automatically reconnect.

closed 2025-06-02 by garthvh

jurgen2005, 2025-08-01:
Same here on the Tbeam het Heltec v3. It used to reconnect right away after signal loss.