meshtastic/Meshtastic-Apple #1288: 🚀 [Feature Request]: change padlock icon for public channels

NomDeTom, 2025-06-22:
### OS

iOS

### Description

Please can the icon for the LongFast channel, and perhaps any other preset defaults, be changed from a red strike-through padlock to something less abrupt?

I would suggest a yellow unlocked padlock or similar.

The reason I'm advocating this change is because:
1) I don't believe the icon is conveying the same message as the same icon for a specific node - in that case, the message is that a fault has occurred.
2) The message to be conveyed here is that the channel is insecure *by design*. The insecurity is a feature not a fault.
3) By using a red icon (which might convey a fault condition in other circumstances), new users are steered away from the default channels, where they simply need to be aware of the inherent lack of privacy it has.

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RCGV1, 2025-06-22:
#1287

garthvh, 2025-06-24:
[image]

Some updates have been made, the red is now reserved for channels pushing a precise location, and there is an extra warning if this is being pushed to the internet.

The lock has been updated to be the open lock and not the one with the slash.

closed 2025-06-24 by garthvh