meshtastic/Meshtastic-Apple #1295: 🐞 [Bug]: received a DM but not shown on any node

holgerscherer, 2025-06-28:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.13

### What did you do?

received a DM from a known node. App shows on messages pane for direct messages a number (ex. 2) but no contact contains any message.

### Expected Behavior

click on a node to display direct messages. also expected to have nodes with unread DM to be sorted on top.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

holgerscherer, 2025-06-28:
addendum: MacOS App Version 2.6.9 (1479)

garthvh, 2025-07-16:
Duplicate of https://github.com/meshtastic/Meshtastic-Apple/issues/1229

closed 2025-07-16 by garthvh

holgerscherer, 2025-07-25:
well this is not fixed yet (for all cirumstances).