meshtastic/Meshtastic-Apple #1305: 🐞 [Bug]: Message Lists (Channel & DM) always start at the top

Xaositek, 2025-07-08:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.7.2

### What did you do?

I have observed that Channel and DM message lists are defaulting to the very beginning of the chat, even if scrolled to the bottom or using the "Jump to latest unread", it still reverts to the beginning of the list

### Expected Behavior

The app will jump to last read message or to the newest message

### Current Behavior

I have observed that Channel and DM message lists are defaulting to the very beginning of the chat, even if scrolled to the bottom or using the "Jump to latest unread", it still reverts to the beginning of the list

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-07-15:
I am not seeing this behavior, both the jump to latest and scrolling are working with several different devices here, do you have bad time maybe?

chase9, 2025-07-19:
I've been seeing this issue as well and was planning on taking a look at it. I think a quick workaround would be to make it so that there is an easy way to scroll to the bottom of the messages list.

chase9, 2025-07-27:
I'm thinking the issue has to do with not properly tracking which messages are unread. It might be helpful to add an unread indicator to the UI so it's easy to see which messages the app thinks are unread.

Xaositek, 2025-09-04:
Despite being closed, this was fixed in the last couple TestFlight builds.

garthvh, 2026-05-17:
This has been fixed in the main branch. Both `ChannelMessageList.swift` and `UserMessageList.swift` now apply `.defaultScrollAnchor(.bottom)` to keep the list anchored to the newest message. On iOS 18+ / macOS 15+ they also apply `.defaultScrollAnchor(.bottom, for: .sizeChanges)` (via the `defaultScrollAnchorBottomSizeChanges()` modifier) so the anchor is maintained when the keyboard appears or the view resizes.