meshtastic/Meshtastic-Apple #1310: 🐞 [Bug]: Create channel then click cancel, still creates channel

va3jme, 2025-07-14:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.11

### What did you do?

Go into channels, create channel, but select cancel as if you decided not to create channel. 
On "Messages" screen, there will be a new channel. On "Channels" there will not be

### Expected Behavior

Channel should not be created. See attachments.

[image]

[image]

### Current Behavior

Go into channels, create channel, but select cancel as if you decided not to create channel. 
On "Messages" screen, there will be a new channel. On "Channels" there will not be

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garthvh, 2025-07-15:
There is no cancel button on this view

garthvh, 2025-07-16:
I can't replicate this, to remove a channel you save it with a disabled role.

closed 2025-07-16 by garthvh