meshtastic/Meshtastic-Apple #1352: 🚀 [Feature Request]: conection via ip on ios

Marlontebaldi, 2025-08-15:
### OS

iOS

### Description

Is there any projection of making connection via IP on iOS?

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pelgraine, 2025-08-16:
Just chiming in to let the devs know I'd be willing to throw in a few dollars to help sponsor this enhancement, and I'm sure others would as well.

ices-dev, 2025-08-25:
This is really useful. I would love to use my Mac to administrate my Linux Native Node without the Web UI :)

jkrauska, 2025-08-31:
duplicate of #1367

closed 2025-08-31 by garthvh