meshtastic/Meshtastic-Apple #1356: 🚀 [Feature Request]: Make it compatible with IOS version 16.7.11

Nullvoid3771, 2025-08-27:
### OS

iOS

### Description

As the title says make it compatible with IOS version 16.7.11, I'm not sure if this version of ios has actual technical limitations. Would prefer not to jailbrake my iphone to find out. Would appreciate the ability it use the app instead of using the web client. 

### Participation

- [ ] I am willing to pay to sponsor this feature.
- [ ] I am willing to submit a pull request for this feature.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

b8b8, 2025-08-27:
https://github.com/meshtastic/Meshtastic-Apple/issues/1199

Nullvoid3771, 2025-08-27:
> #1199

Understood. Was worth a shot.

closed 2025-08-27 by Nullvoid3771