meshtastic/Meshtastic-Apple #1402: 🐞 [Bug]: iOS app stalls BLE→MQTT forwarding in background; only resumes on foreground (needs CoreBluetooth restoration + background reconnect)

shinteza, 2025-09-14:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.7.1

### What did you do?

## What did you do?

**Environment**
- iPhone 14 (latest iOS)
- Heltec T114 v2 (BLE-only, no Wi‑Fi)
- Meshtastic firmware 2.7.7
- Meshtastic iOS app 2.7.1
- MQTT broker: Mosquitto (TLS optional)

**Steps**
1. On the node, enabled **MQTT uplink** on the active channel (standard public channel; no changes to downlink).
2. In the iOS app, enabled **Client Proxy** and configured the MQTT broker (host/port, credentials; TLS optional).
3. Verified in **foreground** that telemetry/messages from the node appear on the broker’s topics.
4. Ensured iOS settings to favor background operation:
   - **Background App Refresh**: ON (for Meshtastic)
   - **Low Power Mode**: OFF
   - **Bluetooth**: ON
   - **Location**: *Always Allow* (where applicable)
   - Did **not** force‑quit the app (left it in the app switcher).
5. Sent the app to **background** and waited ~5–30+ minutes (also tried 60–120 minutes).
6. Monitored MQTT topics on the broker during the background period.
7. Brought the app back to **foreground** and observed MQTT again.

**Result observed**
- Topics updated normally while in foreground.
- After a period in background, topics went **quiet**.
- Upon foregrounding the app, updates **resumed immediately**.

### Expected Behavior

## Expected Behaviour

- With **Client Proxy** enabled, the app should—within iOS constraints—**best‑effort** maintain gateway functionality while in background:
  - iOS should **wake/restore** the BLE central (CoreBluetooth state preservation/restoration) when the peripheral advertises or a relevant event occurs.
  - The app should **re‑attach** to the peripheral and **reconnect** the MQTT client opportunistically on background wakes (BLE restoration, location wake, BG refresh).
- After a background wake, MQTT should **re‑subscribe** and begin forwarding again **without** requiring manual foregrounding.
- **Reconnect latency** after a wake should be reasonably small (target ≤ ~30s).
- For background mode, MQTT config should favor **session persistence** (e.g., `cleanSession = false`) and a **tighter keepalive** to detect stale links faster.
- No promise of continuous real‑time streaming in background (iOS throttles), but forwarding should **resume automatically** when the system does wake the app.

### Current Behavior

## Current behaviour

- After the app sits in **background** for ~5–30+ minutes (sometimes longer), BLE→MQTT forwarding **stalls**—the broker receives **no new messages**.
- When the app is brought back to **foreground**, forwarding **resumes immediately** and the broker starts receiving messages again.
- This is **consistently reproducible** with:
  - Background App Refresh: ON
  - Low Power Mode: OFF
  - Location: Always Allow
  - App **not** force‑quit (remains in switcher)
- Broker logs show a **gap** that corresponds to the background interval, ending right when the app is foregrounded.
- Practical impact: the iOS Client Proxy behaves like a **foreground‑only gateway**, which breaks dashboards/automations unless the app is kept open.

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

NotSwedishBacon, 2025-09-18:
Just a thought, has background app refresh been accidentally removed? Meshtastic doesn’t appear in the list but I can’t confirm if it was there before

shinteza, 2025-09-18:
> Do you have a working branch for this or is it purely suggestions from Gemini?
> 
> Not sure state restore really helps for meshtastic where the state is really stored on the accessory.

I’m sharing what I tested and why state restoration is the lever that actually moves this.

This isn’t about restoring mesh state; it’s about letting iOS wake our **central** after the app gets parked or killed. After a while in the background, iOS freezes the process and may reclaim it, so CocoaMQTT’s auto-reconnect never runs (no process). With a `CBCentralManagerOptionRestoreIdentifierKey` plus a real `willRestoreState`, iOS can relaunch us in the backgroud when it has BLE context to hand back (known peripherals / pending connects / scans). That’s when I re-attach to the node and kick MQTT — no foregorunding needed.

My setup: iPhone 14 on iOS 26, Heltec T114 v2 (BLE-only), FW 2.7.7, Meshtastic app. With state restore  on and a tiny “background gateway” coordinator, here’s what I see:

* Leave the app in the switcher for 30–90 min
* We get a  wake via `willRestoreState` (sometimes a location tick)
* I reconnect to the peripheral (or resume a scan), then call a guarded `reconnectIfNeeded()` for MQTT
* MQTT connects, re-subsribes, and broker topics start moving while the app stays in the background
* Before these changes, broker traffic was silent for the same interval and only resumes the moment I bring the app to foreground
 
Changes in the proof of concept:

* `CBCentralManager` is created with a  restore ID. In `willRestoreState` I reconect handed-back peripherals or restart a filtered scan for the mesh service UUID. No mesh config is “restored” here — just central intent.
* When “Background Gateway” is ON: `cleanSession = false` (subs survive reconnects), `keepAlive ~ 30` (faster stale detection in bg), plus a small backoff to avoid spin.
* Opportunistic nudges: BLE restore callback, a location update (we already have Always), and a BG app refresh task. All three call the same guarded `reconnectIfNeeded()`.
* Info.plist adds `bluetooth-central`, `location`, and `fetch`. It’s opt-in, so default users don’t pay a battery tax

On “the state is on the accessory, so restore won’t help”: agreed the mesh/session lives on the node. The real choke point is iOS putting the app to sleep. Once the OS parks or kills us, nothing reaches MQTT until we use one of the few background wake-ups apple allows and rebuild central + MQTT. State restore is the normal way apple expects a central-role app to get that background entry point. Without it, you wait for the user.

What I saw in practice:

* Battery: negligible in my runs (no long scans; reconnect, re-sub, back to sleep). `keepAlive = 30` vs `60` didn’t move the needle.
* Reliability: still “best effort” (this is iOS, not a daemonn), but noticeably better than “never until foreground.”

If needed, this can sit behind a “Background Gateway (beta)” toggle with guarded code paths only when enabled. The diff is clean (BLE restore wiring, small BG coordinator, MQTT config shim, and OSLog to line up Console logs with broker timestamps).

garthvh, 2025-09-19:
> My setup: iPhone 14 on iOS 26, Heltec T114 v2 (BLE-only), FW 2.7.7, Meshtastic app. With state restore on and a tiny “background gateway” coordinator, here’s what I see:

Is there a working branch somewhere?  This all looks like Gemini suggestions as is.

garthvh, 2025-09-19:
> Just a thought, has background app refresh been accidentally removed? Meshtastic doesn’t appear in the list but I can’t confirm if it was there before

That setting if for pulling down web content in the background on a schedule, meshtastic has never appeared in that list.

closed 2025-10-06 by garthvh