meshtastic/Meshtastic-Apple #1418: 🐞 [Bug]: Last heard date in the future

ejcupcake, 2025-09-22:
### Before submitting

- [x] I have searched existing issues to make sure this bug hasn't already been reported
- [x] I have updated to the latest version of the software to verify the issue still exists

### Firmware Version

2.6.11

### What did you do?

[image]

Some MQTT Nodes have the last heard date in the future. iOS 26.0 w/ App ver 2.7.3

### Expected Behavior

Last heard time should always be in the past

### Current Behavior

last heard time is in the future

### Participation

- [ ] I am willing to submit a pull request for this issue.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

j0uni, 2025-10-06:
I've seen this same thing happening couple of times. 

I've seen situation where "last heard" is 1 month and first heard is "5 seconds ago".

RedstoneGithub, 2025-10-07:
This happens to me frequently, other nodes show Last Heard as the current time or sometime in the future, on first connect to my node. I have to reconnect my node to get the right Last Heard times. Using a T114

closed 2026-05-17 by garthvh